stevemarple/SoftWire #2: Will this work as a replacement for the Wiring library?

smithbp10, 2016-02-05:
There is a Wiring library that is referenced for a specific board (Adafruit PCA 9685) I am trying to use, http://wiring.org.co/reference/libraries/Wire/index.html

Will  SoftWire act as a replacement for that library? That is, assuming I change a few minor lines within that library, such as Wire.h --> SoftWire.h. That library is a pain because it does not easily import into the Arduino native environment, instead the "library" actually implements its own IDE on top of the Arduino IDE.

stevemarple, 2016-02-05:
Wiring is the forerunner to the Arduino environment. The Wire class which you reference is also available in the Arduino environment. Both of these classes use the built-in microcontroller hardware to communicate over I2C, using dedicated pins. SoftWire was created to allow multiple, independent I2C buses by emulating the correct behaviour in software on arbitrary pins. Unless you know that you need a software implementation SoftWire is probably not what you are looking for. I'm not familiar with the PCA9685 but I think what you probably need is https://github.com/adafruit/Adafruit-PWM-Servo-Driver-Library

Amajad, 2017-06-04:
Dear Steve, Could you provide documentation on Softwire library which demonstrate use for multiple i2c bus for reference? any example for multiple i2c bus on different pins?

stevemarple, 2017-06-04:
@Amajad take a look at the example at https://github.com/stevemarple/SoftWire/blob/master/examples/ListDevices/ListDevices.ino

For the second bus (let's call it sw2) using pins 10 and 11 then you'd add 

    SoftWire sw2(SDA, SCL);

Then just duplicate all the functions calls on sw to be called on sw2 also.

Amajad, 2017-06-05:
Thank you sir.

Koepel, 2017-06-05:
Does your library support multiple I2C busses with shared SCL, to reduce pins ?
(if there are not too many sensors or too long wires or too many pullup resistors for SCL).

```
Software sw1( 9, 8);   // sda, scl
Software sw2( 10, 8);
Software sw3( 11, 8);
Software sw4( 12, 8);
```

stevemarple, 2017-06-05:
@Koepel I don't think you can share SCL across I2C busses regardless of the implementation (you'd be sending data to all busses). I think it should be possible to share SDA across I2C busses if all devices are fully compliant. The library wasn't written with that mode of operation in mind but I can't think of anything which would prevent it working.

Koepel, 2017-06-06:
I also could not found something that would prevent it to work.

The SCL should be shared, not SDA. When the SDA goes low, it is the beginning of a start condition. As long as the SDA stays high, the SCL can become high and low, it doesn't matter. The hardware start logic in the slave is never activated when the SDA stays high.

When there are for example ten I2C busses with a single SCL. Then nine of them keep the SDA high, and only one SDA works together with SCL for a start condition, data transfer and stop.

stevemarple, 2017-06-06:
@Koepel It's a while since I wrote the library so I don't remember all the details of I2C but the `stop()` function does set SDA high so I think it will probably work. Please try and let me know how you get on.

Koepel, 2017-06-06:
Sorry, I don't have the time to try it with your library. The shared SCL is a normal thing to do as far as I know.

closed 2017-06-06 by stevemarple