stevemarple/SoftWire #5: Can't compile example file

sumeet33, 2017-11-16:
Hi Steve,

I am using this library to run on LoLin NODEMCU (ESP8266) which doesn't have hardware I2C.
When I try to compile example SoftWire_MLX90614 file, i get following compilation error.

Arduino: 1.8.5 (Windows 10), Board: "NodeMCU 0.9 (ESP-12 Module), 80 MHz, 115200, 4M (3M SPIFFS)"

Build options changed, rebuilding all
C:\Users\sumeetpatil\Documents\Arduino\libraries\SoftWire\src\SoftWire.cpp:1:25: fatal error: util/atomic.h: No such file or directory

 #include <util/atomic.h>

                         ^

compilation terminated.

exit status 1
Error compiling for board NodeMCU 0.9 (ESP-12 Module).

Question:
Can the clock frequency set to 400KHz in this Library?

Regards,
Sumeet

stevemarple, 2018-04-25:
I think the latest version (v1.1.0) will now compile on non-AVR architectures (I only use AVR). Be sure to understand the warnings given in the commit message for b6d873cfbc65502036b5295feee4a464a0c9cc85.

The clock 'frequency' is determined by the delay configured in the class, use `setDelay_us()` to adjust. This doesn't take into account the other delays, such as switching pin mode. For higher speed you can define your own optimized functions which do this (update the pointers `_setSdaLow`, `_setSdaHigh`, `_setSclLow`, `_setSclHigh`, `_readSda`, `_readScl` with the addresses of your own  functions).

Please let me know if this fixes the compile problem for ESP8266.

stevemarple, 2018-05-08:
I've made a minor change to the `SoftWire_MLX90614` so that it now compiles for a generic ESP8266 module. The change can be found in v1.1.1.

closed 2018-05-08 by stevemarple