stevemarple/SoftWire #8: AS3935 library vs SoftWire question

CBMalloch, 2018-08-06:
Recompiled a slightly-modified, but working, program which used AS3935 and SoftWire to speak to an AS3935 in a MOD-1016 card. Wouldn't work. Backed out of changes. Still won't work. Tried demo. Wouldn't work. Hardware not changed.
Signal analyzer sees writes to 0x06 (AS3935 address is 0x03, so this is right) and 0x07, reads 0x24 always. It looks like SoftWire llwrite function buffers bytes to be sent, but nobody seems to call endTransmissionInner, which looks like it would be the thing to actually transmit what's in the buffer!
Here's the sequence I see:
SW: llWrite: 0x6
SW: llWrite: 0x6
SW: llWrite: 0x6
sPU=>sI1
SW: llWrite: 0x6
SW: llWrite: 0x6
SW: llWrite: 0x7
SW: llRead: got 0x24
SW: llWrite: 0x6
sI1=>sI2
SW: llWrite: 0x6
SW: llWrite: 0x7
SW: llRead: got 0x24
SW: llWrite: 0x6
SW: llWrite: 0x6
SW: llWrite: 0x7
SW: llRead: got 0x24
sI2=>sL
SW: llWrite: 0x6
SW: llWrite: 0x7
SW: llRead: got 0x24
SW: llWrite: 0x6
SW: llWrite: 0x6
SW: llWrite: 0x7
SW: llRead: got 0x24
Problem - noise floor was set to 0 but didn't set ==> 2

Did this get broken with the update to SoftWire? Or am I missing something else?

Here's my Arduino code:

#define PROGNAME "MOD-1016_AS3935_lightning_sensor.ino"
#define VERSION "0.3.4"
#define VERDATE "2018-08-04"

/*
    Side notes: that program uses SoftWire, which initializes pin 14 (A0) for SDA
    and pin 17 (A3) for SCL.

  Many "Something else happened!" events
  
  Two kinds of calls - process and getTriggered. What events lead to each?
    getTriggered returns the value of _triggered
    process turns _triggered off if state is WaitR
  Does calling getInterruptFlags and getState clear these? No.
    When should they be called?
  Add a real-time clock to this setup, then hard-wire it; move it to an ESP so can 
    query a timeserver?
    
  v0.3.0 removed dumping registers unless VERBOSE or actual lightning detected
  
*/

#include <AsyncDelay.h>
#include <SoftWire.h>
#include <AS3935.h>

#ifdef JTD
  #include <DisableJTAG.h>
#endif

//************************
//************************

#define VERBOSE 4

const unsigned long BAUDRATE = 115200;
[... 321 lines cut ...]
    Serial.print ( "  Energy of the single lightning: " ); Serial.println ( energy );
  
    if ( energy > 0 ) {
      as3935.readRegister ( 7 , val );
      val = ( val >> 0 ) & 0x3f;
      if ( val < 0x3f ) for ( int i = 0; i < tabColumn; i++ ) Serial.print ( " " );
      Serial.print ( "  Distance estimate, km ( 63 ): " );
      Serial.print ( val );
    }
    Serial.println ();
    
  }
  

  /*
    // no referent!
    Serial.print("EIFR: ");
    Serial.print ( EIFR < 0x10 ? "0x0" : "0x" );
    Serial.println(EIFR, HEX);
  */
  Serial.flush();
  
  firstTimeP = false;
    
}

stevemarple, 2018-08-07:
I'll try to take a look at this later in the week. I now have some hardware to spy on the I2C communications so I'll use that to check that `SoftWire` is sending the expected data. I never had much luck with the AS3935; my breakout board never generated an interrupt.

@CBMalloch Please can you attach that Arduino script instead of pasting it inline. I can't guarantee it will work with my hardware but it might be a useful starting point.

stevemarple, 2018-08-12:
I've not been able to look at the hardware signals but I did see that the `AS3935` library had not been updated to work with `SoftWire` v2.0.0. I've updated `AS3935` to use `llWrite()` instead of `write()` and I believe that v1.0.2 of `AS3935` will fix the problem. Please let me either way.

CBMalloch, 2018-08-17:
It looks like it works fine now. Thanks!

stevemarple, 2018-08-17:
Great - thanks for letting me know so that I can close this issue.

closed 2018-08-17 by stevemarple