stevemarple/SoftWire #9: stm32duino support

pacman1945, 2020-01-23:
I'm trying to use this SoftWire with stm32duino [link](https://github.com/stm32duino/Arduino_Core_STM32).  The library compiles fine and the example ListDevices.ino is working.  However, the read() function seems to be broken on stm32 as it returns incorrect values.  Any ideas how to get this library to work with stm32duino?

stevemarple, 2020-01-23:
At the top of `SoftWire.cpp` this comment explains a potential problem with interrupts:

```
// If possible disable interrupts whilst switching pin direction. Sadly
// there is no generic Arduino function to read the current interrupt
// status, only to enable and disable interrupts.  As a result the
// protection against spurious signals on the I2C bus is only available
// for AVR architectures where ATOMIC_BLOCK is defined.
```

[This](https://github.com/rogerclarkmelbourne/Arduino_STM32/issues/403) thread suggests there may be an atomic block implementation for the STM32 that could be used. 
However I don't know that interrupts are the cause - what happens if you disable interrupts before `read()` and enable them afterwards?

pacman1945, 2020-01-24:
I've tried disable interrupts in llRead(), but the problem persisted.  I've tested another software I2C library, SWire [link](https://github.com/RaemondBW/SWire).  It works fine with stm32duino and didn't disable interrupts.  Thus, I believe the problem with SoftWire isn't related to interrupts.

stevemarple, 2020-09-27:
Did you allocate buffers and pass them into `SoftWire` using the `setTxBuffer()` and `setRxBuffer()` functions? A newly-added example, `ReadDS1307` demonstrates this.

LucasFeliciano21, 2020-10-22:
Hey @stevemarple, I'm also testing your library and I can get it to work for the stm32duino witch is this one [Arduino Core Stm32](https://github.com/stm32duino/Arduino_Core_STM32), and not the one that you've published. I'm not sure if @pacman1945 is looking for the SoftWire for the core that you mentioned because it's native and works pretty well. 

But returning to the question, I've tested with the code for detecting devices, the problem is, there is no response from the devices, the data coming out is always 0, when I change the library to this one [SoftWire from Arduino STM32](https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F1/libraries/Wire/SoftWire.cpp).

Do you have any idea what that might be?

iLoveAndyBaker, 2022-03-18:
I was just having terrible problems with this library on STM32 (STM32F103C8T6), needing a logic analyzer to ferret out the problem, and also a code problem, which was working fine when using the wire library.

The job was a reverse engineering on some Alibaba Ali Express mystery boards, a port from a SAMD / Arduino Zero to STM, and the schematic was botched, requiring software I2C. Doing this work in Platform.io, with soft input pullups enabled.  The chip I'm developing on seems to be a legit STM32F103C8T6, I can confirm the same behavior on clearly counterfeit chips as well.  Literally spent 5 weeks banging my head against this problem. I could write registers, but couldn't read them. 

This may not be the same problem exactly, but the chip and symptoms are. 

This is the change that made it live.
```
iTwoSea.beginTransmission(DevAddr);
iTwoSea.write(RegAddr);
iTwoSea.endTransmission();
//iTwoSea.endTransmission(false); // Original code - This line screws up soft I2C
iTwoSea.requestFrom(DevAddr, (uint8_t)4);
```
The SDA line wasn't dropping to switch to receive.  Getting rid of (false) fixed it. 

[image]
Not what I asked for...

[image]
For the win!