stevemarple/SoftWire #18: Adafruit_MLX90614 and MAX30102 Not working

fervidautomation, 2020-11-20:
Hello Sir,  thank you for sharing such excellent work.

I tried to use ESP32 wroom dev board and use Adafruit_MLX90614 and MAX30102 sensor on I2C communication. As you might know both works on I2C communication. but It did not work. Any piece of advice will be highly appreciated.

stevemarple, 2020-11-20:
At this stage the only advice I can give is to submit more detailed error reports. Wiring details, pins used etc are all useful and sometimes necessary information. The Adafruit_MLX90614 library uses the Wire (hardware I2C) library. I have a library (https://github.com/stevemarple/MLX90614) that supports the MLX90614 and it uses `SoftWire`.

fervidautomation, 2020-11-20:
Thank you for your response. Please find my code below.
I am getting the reading for MAX30102 only but not for MLX90614. both are working on I2C. I did change pin for MAX30102 but still, code is not working.

#include <Wire.h>
#include "MAX30105.h"
#include <Adafruit_MLX90614.h>
MAX30105 particleSensor; // initialize MAX30102 with I2C

TwoWire I2CBME = TwoWire(0);
#define IR1 0x5A
#define I2C_SDA 33  //21
#define I2C_SCL 32  //22
Adafruit_MLX90614 mlx = Adafruit_MLX90614(IR1);
void setup()
{
  Serial.begin(115200);
   while(!Serial); //We must wait for Teensy to come online
 mlx.begin();
   I2CBME.begin(I2C_SDA, I2C_SCL, I2C_SPEED_FAST);
   
  delay(100);
  Serial.println("");
  Serial.println("MAX30102");
  Serial.println("");
  delay(100);
  // Initialize sensor
//  if (particleSensor.begin(Wire, I2C_SPEED_FAST) == false) //Use default I2C port, 400kHz speed
 if (!particleSensor.begin(I2CBME, I2C_SPEED_FAST ,0x57))
  {
    Serial.println("MAX30102 was not found. Please check wiring/power. ");
    while (1);
  }

  byte ledBrightness = 70; //Options: 0=Off to 255=50mA
  byte sampleAverage = 1; //Options: 1, 2, 4, 8, 16, 32
  byte ledMode = 2; //Options: 1 = Red only, 2 = Red + IR, 3 = Red + IR + Green
  int sampleRate = 400; //Options: 50, 100, 200, 400, 800, 1000, 1600, 3200
  int pulseWidth = 69; //Options: 69, 118, 215, 411
  int adcRange = 16384; //Options: 2048, 4096, 8192, 16384

  particleSensor.setup(ledBrightness, sampleAverage, ledMode, sampleRate, pulseWidth, adcRange); //Configure sensor with these settings
}

void loop() {
  particleSensor.check(); //Check the sensor
  while (particleSensor.available()) {
      // read stored IR
      Serial.print(particleSensor.getFIFOIR());
      Serial.print(",");
      // read stored red
      Serial.println(particleSensor.getFIFORed());
      // read next set of samples
      particleSensor.nextSample();    
      delay(1000);  
  }
  Serial.print("Ambient = ");
Serial.print(mlx.readAmbientTempC());
Serial.print("*C\tObject = ");
Serial.print(mlx.readObjectTempC());
Serial.println("*C");
Serial.print("Ambient = ");
Serial.print(mlx.readAmbientTempF());
Serial.print("*F\tObject = ");
Serial.print(mlx.readObjectTempF());
Serial.println("*F");
Serial.println();
delay(1000);
}

stevemarple, 2020-11-20:
The code you posted uses the `Wire` library. Does `MAX30105.h` include or somehow use `SoftWire`?

fervidautomation, 2020-11-20:
I am stuck into this problem to use these 2 sensor on esp32, does softwire, can help me?

stevemarple, 2020-11-21:
So you aren't using SoftWire? Please report any software bugs you find to the appropriate place. Do not use abuse issue trackers as a personal helpline. The majority of authors of open-source software contribute their time for free.

closed 2020-11-21 by stevemarple