stevemarple/SoftWire #21: calling setClock(0) made arduino micro unrecognizable by windows

John-WL, 2021-04-21:
I don't have the source code anymore sadly, and I am kind of scared to repreduce the results on another board. 
The code was very simple, though. It was something like that:

```
#include "SoftWire.h"

SoftWire Wire2(4, 5);

void setup()
{
    Wire2.begin();
    Wire2.setClock(0);
}

void loop()
{}
```

I'm not sure if I called ```setClock(0)``` before or after the ```begin()``` call. 

I'm pretty sure that's what broke it because the parameter ```0``` in the ```setClock(0)``` function is the only value I changed before and after compiling and uploading. 

If it's not only my hardware that's at risk, maybe it would be a good idea to prevent users from setting the clock to 0Hz, or to negative numbers (assuming that's even possible).

stevemarple, 2021-08-25:
Can you provide any information about what you mean by "broke"? Permanently? It required a power cycle? What board and microcontroller were you using?

stevemarple, 2022-04-07:
The bootloader should be protected so that user code cannot mess with it. AVRs have protection for this. If bad user code can mess with the bootloader the error is with fuses not being set correctly.

Since at least 2018-06-02 `setClock()` has had limits to keep the clock period >= 2 µs and 510 µs (ie clock frequency must be in range 1960 Hz to 500 kHz).

I cannot see how the error is due to the `SoftWire` library.

John-WL, 2022-04-07:
Very interesting.
The issue must be something else on my side, then.
I cannot see how it can be something else, but that's my own problem :p
Thanks for the update