stevemarple/SoftWire #22: Restart condition is not issued

hideakitai, 2021-06-23:
Hi, thank you for the great work. I was trying to read the data with a restart condition like this.

```c++
uint8_t readByte(uint8_t address, uint8_t subAddress, SoftWire& wire) {
    wire.beginTransmission(address);
    wire.write(subAddress);
    wire.endTransmission(false);

    wire.requestFrom(address, (size_t)1);
    uint8_t data = 0;
    if (wire.available()) data = wire.read();

    return data;
}
```

A correct transaction with the `Wire` library looks like this.

[image]

But it seems that a restart condition is not issued with `SoftWire`. The hardware setup and code are exactly the same except for the use of `SoftWire` instead of `Wire`.

[image]

I've forced to use `repeatedStart` instead of `start` in [Line 449 of `SoftWire.h`](https://github.com/stevemarple/SoftWire/blob/d040b6b29cea4524bbdfe065313c341487edfef0/src/SoftWire.cpp#L449). Then I could issue a restart condition but the data was not correct.

[image]

Do you have any idea to work around this issue?

stevemarple, 2021-06-26:
That looks like a bug as the repeated start condition is not sent. I need to find some time to look into this in more detail and remind myself of what is supposed to happen.

Can you please post a code snippet showing how you call the SoftWire constructor and setup the RX and TX buffers?

Nice logic traces - what did you use to capture them?

hideakitai, 2021-06-26:
Thank you for the reply. I'm using the following setups.

```c++
// in the global scope
SoftWire sw(SDA, SCL);
char swTxBuffer[64];
char swRxBuffer[64];

// in the setup()
sw.setTxBuffer(swTxBuffer, sizeof(swTxBuffer));
sw.setRxBuffer(swRxBuffer, sizeof(swRxBuffer));
sw.setDelay_us(5);
sw.begin();
```

The entire code is [here](https://github.com/hideakitai/MPU9250/blob/master/examples/connection_check_softwire/connection_check_softwire.ino), and you can reproduce the problem by activating [this line](https://github.com/hideakitai/MPU9250/blob/98ca851dba3376e3171b18a9914eece129d16576/examples/connection_check_softwire/connection_check_softwire.ino#L41) instead of the next one.

I'm using [Logic Pro 16](https://www.saleae.com/). It's a super easy and useful one :)

stevemarple, 2022-04-09:
I finally found the time to investigate this properly. I believe it is fixed in v2.0.8. I wasn't able to test with the same I2C device as you, so I used a Microchip MCP79410 real-time clock. This is the logic analyzer trace obtained by writing to and then reading from that device.
[image]

waynepiekarski, 2022-06-22:
I was encountering a similar problem with 2.0.7 getting the WHO_AM_I register from an NXP FXOS8700 sensor, and after applying this patch my problem has been fixed. Thanks for fixing this.

closed 2022-06-22 by stevemarple