stevemarple/SoftWire #25: llStartWait() case nack needs to include break or return

waynepiekarski, 2021-09-14:
In the llStartWait() function, the stop() call for case nack should include either a break; or a return immediately after it.
Currently my compiler is giving "warning: this statement may fall through".
It looks like it currently will call stop() twice, then return timedOut, but not sure what the actual intent is here.

```
SoftWire::result_t SoftWire::llStartWait(uint8_t rawAddr) const
{
        AsyncDelay timeout(_timeout_ms, AsyncDelay::MILLIS);

        while (!timeout.isExpired()) {
                // Force SDA low                                                                                                                                                                                  
                _sdaLow(this);
                delayMicroseconds(_delay_us);

                switch (llWrite(rawAddr)) {
                case ack:
                        return ack;
                case nack:
                        stop();
                default:
                        // timeout, and anything else we don't know about                                                                                                                                         
                        stop();
                        return timedOut;
                }
        }
        return timedOut;
}
```

stevemarple, 2021-09-18:
I believe this error has already been corrected:
https://github.com/stevemarple/SoftWire/commit/0e111150fe7d3d9832f725089d8a822254945046

waynepiekarski, 2021-09-20:
Ah, thanks, you are right, this was fixed in 2.0.5.

The version on Arduino's download manager is still 2.0.4 with the bug, and that is what I was looking at.

Can this update be pushed to Arduino? Thanks.

stevemarple, 2021-09-20:
Updates are pushed automatically when I add a tag to a commit and push to Github. I can see v2.0.5 in the Arduino library manager (tested with 1.18.13). There were some issues with an incorrectly tagged v2.0.6. I have removed that tag and pushed v2.0.7.

closed 2021-09-27 by waynepiekarski