stevemarple/SoftWire #39: Make SoftWare inherit from ArduinoAPI::HardwareI2C to imporve it as a replacement for the TwoWire-class.

Guswib, 2024-12-01:
I think SoftWire is a very useful library, but, I wish it could be a direct replacement of the TwoWire-class.  Forexample, the Stream-class does not have beginTransmission()-function which for I2C communication is import, in contrast to UART-communication.
So, I propose to the following:
Make the SoftWire libary inherit from ArduinoAPI::HardwareI2C
The HardwareI2C inherit from Steam, so the modification of the SoftWire code should be minimal, especially as the names of the function is the same.

Also, the addition of setTimout(time_in_millisecond) and getTimeout() could be a great addition as those are found in the Stream-class.
What do you all think?

felias-fogg, 2024-12-02:
Your proposal does not work because the methods you want to override are not virtual in the Arduino standard library. And it is not advisable to make them virtual because it will lead to slower and bigger code. See the discussion here: https://github.com/Testato/SoftwareWire/issues/28

stevemarple, 2024-12-02:
TwoWire compatibility has been requested before, see #10 which explains why not, and how to adapt driver libraries to work with  SoftWire.

The timeout functions might be possible but I am nor able to investigate that right now.

Guswib, 2024-12-02:
Thanks you for the replies.
After reading #28 and #10, I can understand why Wire.h cannot be replaced and the TwoWire-class should not be inherited. Each TwoWire implementation is defined on the core level, and conists of large non-virtual functions. So, not very practical in order to save space, and overriding compilation paths would be a nightmare. 
However, as they write in #28 5 years ago, having a "TwoWire"-class that just have virtual functions would work. The issue then was that Arduino needed to implement such a class. Well, with the introduction of the ArduinoAPI and the HardwareI2C-class, the proposed solution has arrived. 
https://github.com/arduino/ArduinoCore-API/blob/master/api/HardwareI2C.h
This class just adds a few extra functionalities to the Stream-class.
I checked the SoftwareWire code, and the change to inherit from HardwareI2C would be about 3 lines of code, as almost all functions has the same name and spelling. And based on that it is still inherit the Stream-class, I cannot see why it would not still work with other libraries.

The use of HardwareI2C:
The TwoWire class from RP2040-core has the implementation:
https://github.com/earlephilhower/arduino-pico/blob/master/libraries/Wire/src/Wire.h
The TwoWire class from AVR-core has not yet made the change:
https://github.com/arduino/ArduinoCore-avr/blob/master/libraries/Wire/src/Wire.h
 
So, it think it is worth to visit this issue again.

stevemarple, 2024-12-03:
The functions `getTimeout_ms()` and `setTimeout_ms()` already exist. Given Arduino sometimes uses microseconds and sometimes milliseconds I prefer to add a suffix that indicates the measurement unit.

closed 2024-12-03 by stevemarple